ysc3839/AudioPlaybackConnector #6: Auto-Connect" feature

mkanet, 2020-09-20:
It would be great if there w

PeggyFree, 2021-01-30:
Hello, thank you for this smart app, it's working flawlessly.
I agree whith @mkanet , it would be usefull to provide a command or a way for my automation system to trigger connecting and disconnecting, rather than having to close then open this app.

resom78, 2021-08-31:
Hi,
This app works very well on my computer, but I would love to have the auto connect feature!
Regards

ysc3839, 2021-08-31:
Maybe in the next version. Currently I use system's device picker ui, it's not easy to make "select a device for auto connect". I will rewrite whole ui in Windows 11 style, but I don't have much time now.

rp1231, 2021-10-10:
@mkanet how are you executing the .exe remotely from your phone?
And killing it as well?

mkanet, 2021-10-10:
> @mkanet how are you executing the .exe remotely from your phone? And killing it as well?

I'm using an app called, **[Unified Remote](https://www.unifiedremote.com/)**.  It's the best Windows remote control app for Windows.  It allows you to create Android Widgets that can execute remote Windows command line commands such as executing AudioPlaybackConnector64.exe or killing the the process.  It was a little bit of effort to setup, but we'll worth it.  I liked Unified Remote so much, I ended up purchasing the full version.  I even added widgets for Wake On LAN my PC or shut it down.  I also use Unified Remote's virtual mouse pad as well and can control a bunch of apps on my Windows PC such as Kodi.

Shihabus-Sakib-Rad, 2022-07-15:
Would absolutely love the command line options !
@mkanet Also you can try [KDE Connect](https://kdeconnect.kde.org/) for remotely executing commands.

OlivierMoser, 2023-03-02:
Hi, did you make any change regarding this feature? Do you think it could be feasible?

iAvoe, 2023-08-12:
Auto connect would be very useful for me, so I don't have to go to my PC to select "connect to this device" everytime, eventhough I just need bluetooth audio from 1 device which is my phone.

This probably needs a listener process to detect device change first, then, create an "auto connect to this device" single-select (or multi-select) check box in the device list UI, which allows only 1 device be selected for auto connect by un-selecting the previous choice if the user selects the second one (or let the most recently connected device be the audio source)