ysc3839/AudioPlaybackConnector #7: 您好 大牛 我想將這個編譯作爲c++ dll庫使用 該如何使用這個？

ShiverZm, 2020-09-21:


ysc3839, 2020-09-21:
需要用做什么用途呢？C++ 可以直接调用相关 API，没必要编译为 DLL。

ShiverZm, 2020-09-22:
想作为软件的一部分 传安卓的声音，具体的c++ api 不太清楚 如果可以 还请不吝指教。

ysc3839, 2020-09-22:
是用什么语言开发的呢？

ShiverZm, 2020-09-22:
c/c++

ysc3839, 2020-09-22:
去了解一下 [C++/WinRT](https://docs.microsoft.com/en-us/windows/uwp/cpp-and-winrt-apis/)，再照着这个项目的代码应该很容易实现。

ShiverZm, 2020-09-23:
这个 实际上是只用一个一些控件吗 没有蓝牙上的交互 对吗

ysc3839, 2020-09-23:
是的

closed 2020-09-28 by ShiverZm

ShiverZm, 2020-11-03:
大佬还是想问一下 标准c++调用c++/winrt 你试过吗

ysc3839, 2020-11-03:
C++/WinRT 就是基于标准 C++ 的，clang 也可以使用。

ShiverZm, 2020-12-03:
补充一点 c# form能调用c++/winrt 吗 我用c++/winrt编译了一个dll,win32调用正常 c#form调用 回调会崩溃 一些winrt的api也会崩溃

ysc3839, 2020-12-03:
.Net Framework 好像能直接调用 WinRT API，.Net Core 也有 C#/WinRT 可用。

ysc3839, 2020-12-05:
我自己没有遇到这种问题。

ShiverZm, 2020-12-06:
好吧 我决定用c# uwp写个类库  这样.net framework可以调用吗 有版本要求吗  像co_wait这种协程 c# .net framwork 4.5好像才能用

ysc3839, 2020-12-06:
我不熟悉 C#，不过印象中 .Net Framework 是可以调用 UWP API 的。版本要求我也不知道。

ShiverZm, 2021-01-07:
我晕 我用mfc 调用winrt写的dll 声音连上 就断开了

ShiverZm, 2021-01-07:
大佬有群吗 一起学习呀

ysc3839, 2021-01-07:
@ShiverZm
https://t.me/cpluspluszh
接下来请到这里讨论吧。

ShiverZm, 2021-03-06:
@ysc3839  大牛 我发现一个问题 我之前mfc调用您的connectdevice 会断开，但是我只要把connectDevice内部 sleep 一直在 就能一直连接

ShiverZm, 2021-03-06:
然后想请教关于 监听蓝牙设备连接的 api 请问大佬知道吗 找了很多都是获得连接过的设备列表