ysc3839/AudioPlaybackConnector #10: Bluetooth stutters

rp1231, 2021-01-17:
Having audio stutters while using this.
Happened with the other uwp application that exists.
Do you have any idea how to solve it?

Also, thanks for making this!
It was much needed.

Atrello, 2021-01-31:
I also have this issue, would love a fix as I'm trying to use this to stream from my TV to my PC to use wireless headphones :(

ysc3839, 2021-02-01:
Sorry, I have no idea. This program is just a front-end for the underlying system API. The connection and transmission is handled by the system.

closed 2021-02-01 by ysc3839

mkanet, 2021-02-01:
I wonder what's the difference between those of us that don't have this issue and the ones that do.  I use my Samsung Note 9 to connect to my HTPC/Gaming PC via bluetooth without any stuttering issues when sending audio.  It actually very reliable.

ysc3839, 2021-02-01:
I guess it's hardware issue. Maybe caused by cheap Bluetooth module or bad antenna.

rp1231, 2021-02-01:
Using the intel bluetooth wireless chip that came installed with the laptop.....
Have ordered a usb dongle to see if it makes any difference. Will post my findings here in a few days after it arrives.

Atrello, 2021-02-01:
I use this
https://www.amazon.co.uk/TP-LINK-UB4A-Bluetooth-Computer-Receiver/dp/B07YLDVM6B/
exact model of Bluetooth adapter, I think it must be a system side issue.

rp1231, 2021-02-02:
Switching to a usb dongle solved my stuttering problems.
So I guess the problem was with my laptop's internal bluetooth card after all.

Atrello, 2021-02-02:
Hmm weird, I still get the issue with a Bluetooth dongle.

Atrello, 2021-02-02:
I am on desktop so it's my only option

ShadowFlare, 2021-05-07:
There's likely interference coming from the computer on the 2.4 GHz frequency band, which Bluetooth also uses. Plugging in the receiver with some type of USB extension to move it away from the computer should help considerably, and some receivers (whether Bluetooth or for any other wireless peripheral) even come with some type of extender for this very reason (either just a short extension with no cord or an extension cord).

ShadowFlare, 2021-05-07:
I don't even mean things that use wireless communication are giving off the interference. Some components that don't even have anything to do with anything wireless whatsoever just give off interference in that frequency band or others. USB3 is one known source of interference in the 2.4 GHz band, though there can be any number of other components giving off interference in various frequency bands. Basically any electronic components that are unshielded will give off some form of radio signals. The best way to deal with this is to move the antenna away from the source of interference (the computer), which in this case the antenna is integrated with the receiver in the USB Bluetooth receiver. Moving it away from the computer with a USB extension can significantly reduce Bluetooth lag and stutters and can sometimes improve range substantially.