ysc3839/AudioPlaybackConnector #11: Searching everytime I open the app

rp1231, 2021-02-02:
Even after my phone is connected to the app,
It spends a few seconds searching for devices before showing that the phone is actually connected.
It is not a functionality issue but just a visual improvement that I think would make this app better.

Thanks again for the app.

ysc3839, 2021-02-03:
Sorry, I can't reproduce this issue on my computer. Can you post a video or gif?

rp1231, 2021-02-03:
[image]

Hi,
I hope this gif is clear enough to illustrate the problem.
It's a very minor thing but as you can see the phone had already been connected,
but it still shows the "Make sure the device is connected" message every time I click on the taskbar icon.