ysc3839/AudioPlaybackConnector #15: Paired computer isn't showing in the list

JiaChen-Zeng, 2021-08-20:
I'm trying to use this app to play audio from another computer, but that computer is not showing in the list so I can't connect. I can see my computer is already paired. For my phone, I can connect to it normally. Is playing audio from another computer not supported?

ysc3839, 2021-08-20:
> Is playing audio from another computer not supported?

I think so. I remembered I tested this case some months before and had same issue.

Jearde, 2022-07-07:
I have the same issue with a Google Nest Hub. The Nest Hub is paired, but I not shown in the device list of AudioPlaybackConnector. Will there be support other devices than phones?

nerdamigo, 2022-07-27:
This is a screenshot of the Bluetooth services offered by my phone, which works as a source for this sink.
[image]

Comparatively, when I connect another Windows 10 PC, running 21H2, I only see the following services. While it does include "Audio Source" - the device never shows up in the list of connections in this application.
[image]