ysc3839/AudioPlaybackConnector #18: Not Working In Windows 11

KIANCHONGYT, 2021-12-22:
Hope can fix it and work in Windows 11
Thank You 😊

KarimFereidooni, 2021-12-31:
It works for me in windows 11 😎

KIANCHONGYT, 2021-12-31:
> It works for me in windows 11 😎

I can't I press the taskbar icon it not showing anything I also run it on administrator also didn't work 😔

brentleywilson, 2022-04-19:
There are some problems with WIndows 11 specifically. I dont know if its due to my hardware but the thing actually works. The "Phone Link" app can cause problems. So first stop phone link on your android device if you have it and then pair it with windows, if you get an error stop the AudioPlaybackConnector from task manager and then try. Once paired try connecting from the dialogue of AudioPlaybackConnector and it should work.

utkarshsethi, 2022-09-02:
> Hope can fix it and work in Windows 11 Thank You 😊

Provide details as to what is actually wrong?

> > It works for me in windows 11 😎
> 
> I can't I press the taskbar icon it not showing anything I also run it on administrator also didn't work 😔

Have you check where the shorcut is pointed to? is the exe actually there?

jingcjie, 2022-12-03:
> There are some problems with WIndows 11 specifically. I dont know if its due to my hardware but the thing actually works. The "Phone Link" app can cause problems. So first stop phone link on your [...]

Newest version of win 11. I used the "Phone Link" at the same time, it's totally okay, nothing wrong will happen.
最新的win10没问题，也不会和win自带的连手机app冲突，这个bug怎么复现啊。