ysc3839/AudioPlaybackConnector #23: Anyone know a a solution to Audio Pops/Crackles/Glitches?

hockinsk, 2022-09-03:
I know this isn't an issue caused by AudioPlaybackConnector because the same happens with other A2DP Sink connectors such as the one on Microsoft Store. I've tried a few things. With and without Wifi on, ensuring everything is running at 44100hz 16-bit which his what the Recording A2DP Sink seems to be fixed at. Not having much luck and tried 2two different audio interfaces. Anyone have any advice on fixing the issue? The glitches seem random, perhaps once or twice a minute.

Banana-97, 2022-09-19:
I believe its due to the Bluetooth version/ dongle or your device

mkanet, 2022-09-19:
@hockinsk If you want to avoid this issue and also increase the range between the Bluetooth radio and Bluetooth devices, you should consider getting the below Bluetooth dongle.  It was the best upgrade I made for Bluetooth; and works great with AudioPlaybackConnector.
https://www.amazon.com/gp/product/B09BFGDPVG/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

mbartlett21, 2024-06-20:
This also happens quite often when using AudioPlaybackConnector with bluetooth headphones. :)