ysc3839/AudioPlaybackConnector #28: Select audio output device for sound comming from bluetooth device

K4M1s, 2023-04-09:
It would be awesome if there were an option to select which audio device sound should be outputed to. I've tried changing the output device in audio mixer (windows 11) but it is not working :/

NoPlagiarism, 2023-07-13:
Go to `Sound` -> `Recording` -> there you should find `%your_device% A2DP SNK` -> `Listen`. There you can switch output device

Prabesh01, 2023-07-14:
@NoPlagiarism There's no "A2DP SNK" in my case.
[image]

K4M1s, 2023-07-14:
Can agree, changing output in mixer doesn't work

Peekofwar, 2023-12-17:
I'm going to assume that restarting the program after modifying the output device in Settings doesn't result in anything...?

Frassensh, 2023-12-17:
Unfortunately no, nothing changes. I have made a topic on the microsoft forum about this also, but nobody seems to have an aswer to the problem there either. 

https://answers.microsoft.com/en-us/windows/forum/windows_11-hardware/bluetooth-a2dp-snk-device-not-showing-up-in-sound/1d4e72e1-8f70-4356-9fea-3df5510c32fe

sanecchkarrr, 2024-09-23:
I tried enabling the display through the Windows registry, but it didn’t help. My opinion might be wrong, but I think that Microsoft updated the microsoft_bluetooth_a2dp_snk.inf driver in a Windows update, which hides the device from the Sound panel. I believe it’s possible to extract the old driver from Windows 10 and install it on Windows 11, or continue trying to make changes in the registry.

Frassensh, 2025-01-02:
My thread on microsoft forum finally got an answer that works!
https://answers.microsoft.com/en-us/windows/forum/windows_11-hardware/bluetooth-a2dp-snk-device-not-showing-up-in-sound/1d4e72e1-8f70-4356-9fea-3df5510c32fe

HennieLP:

Found an trick if you go in the registry under Computer\HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\MMDevices\Audio\Capture then you have to find the right device the name of the device can be found under properties and then delete the Key {9c119480-ddc2-4954-a150-5bd240d454ad},6 under properties now it will show up under sound settings and will work for example with Voicemeeter