ysc3839/AudioPlaybackConnector #34: List of devices not appering on windows 2022 server

adamerso, 2023-08-12:
However Microsoft finally added bluetooth to their server line operating systems, there is an issue with your app :(

It simply ignores left mouse click on tray icon, right clicks opens bluetooth connections without an issue. 

Tried with fully installed bluetooth drivers (one from widcomm and one newer from Asus)

Bluetooth hardware - Asus BT400 usb dongle.

Marctraider, 2023-08-13:
Same issue on 21H2 (LTSC).

Marctraider, 2023-08-13:
Maybe it could be a service disabled on server/LTSC, or a missing dependency?

If disabling WLAN autoconfig service, left mouseclick also wont show available wifi AP list.

CledsonVilanculo, 2025-06-14:
its happenig to me too, it used to work, but after a time it started this issue