ysc3839/AudioPlaybackConnector #35: Not connecting anymore on fresh windows 11 install ?

ShySoviet, 2023-09-02:
Hi ! 

First of all thanks for this software it's really good.

I met an issue when trying to use the app on a totally new Windows 11 install. My phone and my PC are paired, but everytime I try to connect, I get the following : "a device attached to the system is not functionning (0x8007001F)". 

I tried reinstalling every bluetooth driver, restart my devices, pair my pc and phone again, do the thingy in Services.msc but nothing worked.

I do remember that one time, on my older installation it suddenly stopped working (same behavior as today) and I managed to fix it with the bluetooth troubleshooter (I don't remember what the utility fixed though...). But with the new version of Windows 11, they introduced a new troubleshooter for bluetooth, and it's really bad, it doesn't find any issue, so I can't do anything...

Do any of you guys know how I could try to fix this issue ? Anything will be appreciated. 

Thanks :)

Gladuin, 2023-09-06:
I have the same error on Windows 10 LTSC, but it's even weirder for me. My old Samsung phone works fine, but my Google Pixel 7 gives that same error.

My old chinese bluetooth dongle DID work fine with both my pixel and my samsung, but that one doesn't play nicely with my other bluetooth devices...

Really weird situation.

Link2006, 2023-10-26:
I've got the same issue but trying to connect my Nintendo switch instead, the other software also just closed when opening a connection, which according to the event viewer, it's a crash.

I am running Windows 11, i also tried on windows 10 and it also fails with the same error