ysc3839/AudioPlaybackConnector #39: Is it possible to capture Sound to OBS or any Stream programs?

kgy9948, 2024-07-21:
"The title is the content"

ysc3839, 2024-07-28:
After connect, there will be a new recording device. Just add that device to OBS or other recording software.

kgy9948, 2024-07-31:
Can't Find "new recording device" like program name or sound device, recording system captures only system sound (like realtek 3.5mm) or programs WITHOUT audioplayback. is there any way to record the bluetooth device sounds? 

[image]
[image]

shugen002, 2024-08-01:
I have same question, I notice there is A2DP SNK device in device manager.
But it didn't show up in system neither recording device setting nor playback device setting.
And also not show up in OBS audio input capture or audio output capture device list.
[image]
[image]
I think this could be a system limition ? (Windows 11 23H2 22631)

ysc3839, 2024-08-03:
@kgy9948 @shugen002 
First connect BT output device in AudioPlaybackConnector.
Then play some music in that output device. (To make recording device active, otherwise it won't show in below step)
Open OBS, click Settings, Audio, select "Line (*Device Name* A2DP SNK)" in Mic/Auxiliary Audio.
Then you can see that device in Audio Mixer, you can now start recording.
[image]