ysc3839/AudioPlaybackConnector #44: Audio comes through, but no ability to play/pause from PC or integration with Win10 OSD?

Wizek, 2025-05-24:
I'm not sure if this is a known limitation, but even though the audio is connected very nicely and comes through crystal clear, I don't see any way to control the playback from the PC on what's happening on the phone. And in other Bluetooth connections this is usually possible with let's say a Bluetooth speaker that usually has play and pause buttons. So not sure if it just is broken for me or if this is a known current accepted limitation of this software. And is there openness to accept contributions and pull requests regarding adding this support one way or another, if this is a known situation?

MadafakinRio, 2025-07-04:
I had the same "idea", it would be nice to be able to do it, although it is a very minor use-case for me. I think that other than supporting AVRCP, the app would also need to hook into the Windows' API to act like a media player app. It's very possible. Btw I sadly think this project is abandoned (seeing as the last commit was 5 years ago), so maybe fork it?

flic1, 2025-08-11:
fyi: on Windows11, you can control playback (prev/next track and pause) from the PC with the Phone Link app
theres even the picture of the track showing

[image]